prusa3d/Prusa3D-Test-Objects #11: 📑Report: Retraction test results

3d-gussner, 2020-06-05:
Please report here your Retraction test results.

Use the 3mf file below and try to pint it with PETG or some other material that tends to be stringy.
[Container_50x20_NoCore_v2.zip](https://github.com/3d-gussner/Prusa-Firmware/files/4585431/Container_50x20_NoCore_v2.zip)

- Printer: MK2.5, MK2.5s, MK3, MK3s
- MMU: MMU1, MMU2, MMU2s
- Firmware version printer and MMU
- Bad to good scale : 0 - 10

Feel free to write additional information you want to share.

I will try to update following table so everyone can see what results we got from the community.

|Printer | MMU | PFW | MFW |K value| Quality | User |
| ------ | ------- | ---------- | ---- | ------- | --------- | -------- |
| MK3s | N/A | 3.8.1 | N/A |66| 2 | Average |
| MK3s | N/A | 3.9.0-RC3 | N/A |66| 0 | Average |
| MK3s | N/A | 3.9.0-RC3 | N/A |0.11| 9 | Average |
|  |  | |  |  |  |  |  |  |  |
| MK3s | N/A | 3.8.1 | N/A |66| 2 |  Vossberger |
| MK3s | N/A| 3.9.0-RC3 | N/A |66| 0 |  Vossberger |
| MK3s | N/A | 3.9.0-RC3 | N/A |0.11| 9| Vossberger |

3d-gussner, 2020-06-05:
**Copy of @Vossberger [results](https://github.com/3d-gussner/Prusa-Firmware/issues/72#issuecomment-625183020)**

(MK3s, no mmu, Ruby Nozzle, SD-Card, Nunus "Transparent hellblau" PETG 

FW3.8.1 K66           very stringy (!), Extrusion up to 160% necessary, Result =2
FW3.9.0-rc3 K66     looks like one complete body, extrusion=100%, Result=0
FW3.9.0-rc3 K0.11  no string, extrusion=100%, Result=9

(Photo from bottom to top : FW3.8.1 / FW3.9.0rc3 LA1.0 / FW3.9.0rc3 LA1.5)
[image]

3d-gussner, 2020-06-05:
**Copy of @wavexx [comment](https://github.com/3d-gussner/Prusa-Firmware/issues/72#issuecomment-625256143)**

Who-ho.. massive difference. I'm missing part of the discussion. How K66 /  K0.11 are chosen?

3d-gussner, 2020-06-05:
**Copy of my [comment](https://github.com/3d-gussner/Prusa-Firmware/issues/72#issuecomment-625797183)**

The LA1.0 to LA1.5 conversion is way off for this filament with K=0.21 vs. real LA15 K=0.11.
So the issue is in the conversion OR in the initial LA1.0 K value
```
Send: M900 K66
Recv: LA10C: Linear Advance mode: 1.0
Recv: LA10C: Adjusted E-Jerk: 4.50
Recv: echo:Advance K=0.21
Recv: ok
```
**Looking at other PETG profiles**, they all have LA1.0 M900 K45 which is very close to the LA1.5 K=0.11
```
[...]
Send: M900 K45
Recv: LA10C: Linear Advance mode: 1.0
Recv: LA10C: Adjusted E-Jerk: 4.50
Recv: echo:Advance K=0.13
Recv: ok
```
@Vossberger Can you please rerun the k-factor for **LA10** for your PETG? I really think there is the problem as all other tests/results with PETG were nearly the same as with FW3.8.1

3d-gussner, 2020-06-05:
** Copy of @Vossberger [comment](https://github.com/3d-gussner/Prusa-Firmware/issues/72#issuecomment-633456270)**

Once again to this test set. It is possible that I did not adjust the eSteps before changing the FW again (I use a 0.9 stepper [415 -> 830].

carcamerarig, 2020-10-04:
Printed test with Extrudr PETG 1.75 usually print at 230 /85 lowered to 220/ 70 for this test.
Struggling with stringing from the point at when the extruder moves away for an octoprint photo and returns.

[Container_50x100_NoCore_v2_0.2mm_PETG_MK3S_1h18m_Octoprint.gcode.zip](https://github.com/prusa3d/Prusa3D-Test-Objects/files/5323559/Container_50x100_NoCore_v2_0.2mm_PETG_MK3S_1h18m_Octoprint.gcode.zip)
[image]
[image]

3d-gussner, 2020-10-05:
Octoprint photo is moving the head aside and taking a photo and continuies to print, correct?
I think this is another issue as the hotend will ooze some filament during this procedure and then kick off these tiny pillars.
You should look for octoprint photo tuning, this is not an issue of the firmware. This is a process issue. First thing I could think of is to use the pause retraction and unretract during the photo pause.

carcamerarig, 2020-10-05:
I have already asked there, but didn't get much response so trying to understand what's happening myself and come up with a solution, it's why I asked here. The photo move would be the same as moving to another print on the bed wouldn't it? What would pausing the retract, unretract do to relieve this? I thought about a small tower i between the travel (is that called wiping?) but the stings are being pulled from the print before it moves first, the ooze then adds to it.
Of course I understand this is an octoprint issue but some thoughts and advice would be appreciated.