prusa3d/Prusa3D-Test-Objects #13: 📑Report: LA1.5 fix

3d-gussner, 2020-06-19:
Please report here your LA1.5 fix test results.

- [G] Check for gaps at seam and other places     _0=big gaps 10=perfect seam_
- [V] Check for vertical lines, good to see at arcs  _0=very        10=not visible_
- [A] Check for overall best result                          _0=bad         10=best_

Use the 3mf and gocde files in zip below.
[Speedtest_simple_25mm_every_5mm_different_speeds.zip](https://github.com/prusa3d/Prusa3D-Test-Objects/files/4803489/Speedtest_simple_25mm_every_5mm_different_speeds.zip)
Every 5mm the speed changes
[image]

- Printer: MK2.5, MK2.5s, MK3, MK3s
- Firmware version printer
- Bad to good scale : 0 - 10

Feel free to write additional information you want to share.

I will try to update following table so everyone can see what results we got from the community.

|Printer | PFW | 25mm/s | 50mm/s | 75mm/s | 100mm/s | 125mm/s | User |
| ------ | ------- | --------- | -------- | ------- | --------- | -------- | -------- |
| MK3s | 3.8.1 | Gxx Vxx Axx | Gxx Vxx Axx | Gxx Vxx Axx | Gxx Vxx Axx | Gxx Vxx Axx | Average |
| MK3s | 3.9.0 | Gxx Vxx Axx | Gxx Vxx Axx | Gxx Vxx Axx  | Gxx Vxx Axx | Gxx Vxx Axx | Average |
| MK3s | 3.9.0+fix | Gxx Vxx Axx | Gxx Vxx Axx | Gxx Vxx Axx  | Gxx Vxx Axx | 50mm/s | Average |
|  |  | |  |  |  |  |  |  |  |

Kachidoki2807, 2020-06-20:
I did a quick test, flashed the 3.9.1-BETA-3459 and printed the G-Code.

No, it's not a multi-color print, only the speed variations:
[image]

I had big gaps @125mm/s:
[image]
[image]

Less visible, but there are also tiny gaps at other speeds:
[image]

The best is 100mm/s I think:
[image]

About the vertical lines, the worst in arcs is 50mm/s, and for the straight lines (not parallel to X or Y) it is 75mm/s.

DRracer, 2020-06-21:
@Kachidoki2807 hmmm, this is weird. I tried the same G-code a few days ago on MK3S and MK3 and none of them had gaps at all. 
I built the FW myself from https://github.com/wavexx/Prusa-Firmware/tree/la15_acc_test
Fillament: random spools of Prusament PLA red.
 
I'll try the FW included in this issue on another MK3S just to make sure.

Kachidoki2807, 2020-06-21:
Okay... I'll give a try with another filament on the same printer, and also on an another printer to see if this behaviour is consistent.

I have to mention that my nozzle must be pretty weared as I printed several kilos of face shields with it. I am waiting a replacement one, but in the meanwhile the print quality seems unaffected.

DRracer, 2020-06-21:
@Kachidoki2807 just got similar gaps as you do (the top layers), running a clean FW build from @wavexx 's repo to isolate the problem (that build was ok for me on another set of printers). The gaps shouldn't be there anymore, something is wrong. My machine is relatively new, roughly 1kg of PLA went through (a bit of ASA and FLEX too).

Kachidoki2807, 2020-06-21:
@DRracer I confirm, other filment on same printer or other filament on other printer, I got the same gaps.

I resliced the project from the .3mf and scaled down the Z axis by half to speed up the test. At the botom is the first purple test,at the middle in brown it is on the same printer, and at the top in red it is on an MMU2S (Sorry for the bad quality picture).
[image]

wavexx, 2020-06-21:
I can reproduce this!

3d-gussner, 2020-06-22:
New fix from @wavexx has been released please try [**FW3.9.1-BETA-3464**](https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.9.1-BETA-3464) 22 June 2019

Kachidoki2807, 2020-06-22:
FW 3.9.1-BETA-3464

Almost perfect:
[image]

Kachidoki2807, 2020-06-22:
> New fix from @wavexx has been released please try [**FW3.9.1-BETA-3464**](https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.9.1-BETA-3464) 22 June 2019

22 June 2020. ;-)

mmuellerphoto, 2020-06-23:
OK ... So here my results:
Zaribo Mk3s 320, Bondtech, Mosquito Magnum, Brass Nozzle, Redline PLA

FW | 25mm/s | 50mm/s | 75mm/s | 100mm/s | 125mm/s | Comment
-- | -- | -- | -- | -- | -- | --
3.8.1-2872 | G10 V04 A04 | G10 V06 A06 | G08 V07 A07 | G07 V03 A03 | G08 V02 A02 | strange horizontal lines at lowest and higher speeds
3.9.0-4153 | G10 V05 A04 | G10 V07 A07 | G07 V10 A09 | G04 V08 A08 | G02 V03 A03 | strange horizontal lines at lowest and higher speeds; visible gaps at sharp corner
3.9.1-4179 | G7 V04 A04 | G8 V07 A07 | G05 V09 A09 | G06 V08 A08 | G02 V02 A03 | strange horizontal lines at lowest and higher speeds; visible gaps at sharp corner
3.9.1-4202 | G5 V09 A09 | G4 V09 A09 | G03 V09 A09 | G06 V08 A08 | G02 V04 A03 | strange horizontal lines at lowest and higher speeds; visible gaps at sharp corner at highest speed

Latest FW brought best results: sharp corners, best surface look. Did not play with temps, so different shades of color on the surface.

The gap / horizontal line on the surface with the 3.9.0 test is related to a filament change because of an empty spool. Same filament used afterwards.

[image]

[image]

[image]

[image]

[image]

Kachidoki2807, 2020-07-15:
Hi,

@wavexx, @3d-gussner, can I consider this topic (gap in corners) solved?
I am waiting that this topic is officially solved to switch my MK3s setup to a plated copper 0.6mm nozzle, then I will not be able to do more tests on this printer in the same conditions.

From my side, I am using the 3.9.1-BETA-3464 since three weeks now, I didn't noticed visible artifacts on the printed parts like I have on the other printer with the stock firmware at high speeds.

Thanks.

3d-gussner, 2020-07-15:
Thanks for the update, we still investigate some rare issue, so it may take a bit until we have fw3.9.1 rc or final.
We might ask you to test with the 0.6mm nozzle, good to know that you have it. Hope that's okay for you.