prusa3d/Prusa3D-Test-Objects #14: 📑Report: Belt test fix results

3d-gussner, 2020-06-19:
With FW 3.9.0 we improved the belt test, but there have been some issue reports:
1. Belt test value changed significant. This is an expected result and we are working on a solution.
2. Belt test randomly creates errors on x and/or y axis.

To improve the shown LCD belt test values (issue 1.) we would like to ask you to report back here which belt test values you get with different firmware versions:

Please also report back if belt test fix is working as we found in internal tests.
- Test at least 20 times
  - Try 10 time with the hotend and bed after homing
  - Try 10 time with random positions of the hotend and bed

Please also add following information:
- Printer: MK2.5, MK2.5s, MK3, MK3s and mods like Bontech extruder and 0.9° motors
- Firmware version printer
Here some links to the Firmware hex files:
[FW3.9.1-BETA-3459](https://github.com/3d-gussner/Prusa-Firmware/releases/tag/v3.9.1-BETA-3459)
[FW3.9.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.9.0)
[FW3.8.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.8.1)

Feel free to write additional information you want to share.

I will try to update following table so everyone can see what results we got from the community.

|Printer | PFW | Belt status x | Belt status y | Belt test errors x | Belt test errors y |  User |
| ------ | ------- | --------- | -------- | ------- | --------- | -------- |
| MK3s | 3.8.1 | 275 | 278 | 0 of 20 | 0 of 20 | Average |
| MK3s | 3.9.0 | 179 | 230 | 4 of 20  | 1 of 20 | Average |
| MK3s | 3.9.1-BETA-3459 | 227 | 303 | 0 of 20  | 0 of 20 | Average |
|  |  | |  |  |  |  |  |  |  |

3d-gussner, 2020-06-19:
Here my results:
Printer: Prusa MK3s stock no mods
FW 3.8.1 x=275 y=278, 0 errors on x and y with 20 tests
FW 3.9.0 x=179 y=230, 4 errors on x and 1 error on y with 20 tests. Got some errors when hotend and bed were at random position (NOT home)
FW 3.9.1-BETA-3459 x=227 y=303,  0 errors on x and y with 20 tests

vintagepc, 2020-06-20:
- 3.9.0-, ~RC3-ish, (added a few features I use myself)
- 0.9 steppers on X/Y
- Nonstock bearings on X/Y

[image]

I can't run 3.8.1 as my printer is particularly prone to https://github.com/prusa3d/Prusa-Firmware/issues/2306

At some point today I hope to have time to merge your tweaks into my local branch and run with those

vintagepc, 2020-06-20:
:thinking: 

Hmm... 3.9.1 patch:
(I stopped early on the random ones because for some reason, despite hearing the steppers shut off they are still locked after belt test... This was reported before but I could not reproduce it at the time. Seems it's back...)

[image]

leptun, 2020-06-20:
I think it’s the stepper inactivity timeout messing with us.

Kachidoki2807, 2020-06-20:
My average results:
Old firmware on which I have done the last Belt Test, don't know the version : X: 281 & Y: 311.
3.9.0 -> X: 254 & Y: 293, but Y failed each times.
3.9.1-BETA-3459 -> X: 279 & Y: 309.

3d-gussner, 2020-06-20:
> My average results:
> Old firmware on which I have done the last Belt Test, don't know the version : X: 281 & Y: 311.
> 3.9.0 -> X: 254 & Y: 293, but Y failed each times.
> 3.9.1-BETA-3459 -> X: 279 & Y: 309.

You had no errors with FW 3.8.1 and FW 3.9.1-BETA-3459
and 20 of 20 on y with FW 3.9.0?

leptun, 2020-06-20:
@3d-gussner the lower belt test current used in 3.9.0 can cause more noise in the endstop data. Sometimes it is more extreme than other times.

vintagepc, 2020-06-20:
> I think it’s the stepper inactivity timeout messing with us.

Eh, I think I did see it's consistent with auto-home (also locked) so I don't see any reason to break from the established behaviour and fix the one case for belt test.

Kachidoki2807, 2020-06-21:
> You had no errors with FW 3.8.1 and FW 3.9.1-BETA-3459
> and 20 of 20 on y with FW 3.9.0?

I confirm, no error at all with FW 3.9.1-BETA-3459. I didn't tested with the FW 3.8.1.
And also I reflashed again the FW 3.9.0, and none has finished, always "Loose pulley" on Y.

I think to have noticed something different between the two firmware, to be checked, I have the feeling that the "knocking" sound is different when the Y plate touch the back of the printer. It seems to be louder, maybe like a multiple knocking.

rcaslis, 2020-06-21:
I tested on 3.8.1 and 3.9.

On 3.8.1, The X values ranged from 234-238 and the Y values ranged from 253-258.

I gave up on 3.9 because I kept getting an X axis length error.  Yes, my x-carriage does have increased resistance on the far edge but it wasn't an issue before.  This is with Vesconite bearings.  I did try and loosen the screws which did permit the belt test to proceed (X values 199-200, Y values 211-214) but honestly the whole extruder was loose so I didn't trust anything I was seeing.

carlin57, 2020-06-23:
I tested on 3.9 and 3.9.1 BETA (mk3S BONTECH extruder)

3.9  5 belt test: 3fails on X , the values for X 195>205 and the Y values 225

3.9.1 belt test: the X values ranged from 218-220 and the Y values ranged from 225-226 ( 5test too) and no fails 
sorry for my english

ochm, 2020-07-14:
Belt test results after firmware change:
3.9.0          246/267
3.9.4.3464 267/280

etix, 2020-07-17:
MK3S (no mod)

**3.8.1**: X: 261 Y: 261
**3.9.0**: X: 250 Y: 274
**3.9.1-3464**: X: 268 Y: 283

mamurluk, 2020-11-01:
I have a self-assembled mk3s bear printer. when belt testing gives an error "please check axis length x". 
the error was on versions 3.8.X and 3.9.X

vintagepc, 2020-11-01:
> I have a self-assembled mk3s bear printer. when belt testing gives an error "please check axis length x".
> the error was on versions 3.8.X and 3.9.X

Neither of the places you've posted this are support channels for resolving that problem, they are developer channels for working with the measurement results. 

In your case the issue is as described, the measured X length differs from what the printer expects. Check for loose pulleys, etc. The proper support channel is the Prusa live chat and/or the forums.