prusa3d/Prusa3D-Test-Objects #15: 📑Report: E-Jerk: diagonal lines and zig zag

3d-gussner, 2020-07-10:
Please print the following simple square. It takes around 25min and prints at 125mm/s.
The PLA temp is set to 235°C but to get more shiny/transparent walls you can crank it up to 270°C at these speeds.

[image]

[Thin_Square_LA15_E_jerk_4_45_to_4_55_PLA.zip](https://github.com/prusa3d/Prusa3D-Test-Objects/files/4902850/Thin_Square_LA15_E_jerk_4_45_to_4_55_PLA.zip)
The E-jerk from bottom to top starts with 4.45 and increases by 0.01 every 5mm to 4.55.

We are looking for diagonal lines and zig zag pattern at the walls.
Report back which E-jerk section looks the best.
Diagonal lines zig zag : 0 = worse 10 = best

Example: It is hard to see that iss why i tried to "paint" the visible zig zag and diagonal lines 
[image]

If you have the time try different FW but the latest https://github.com/wavexx/Prusa-Firmware/tree/la15_acc_untested is also ok

| PFW | E-jerk | Diagonal lines | zig zag front | zig zag side | zig zag back |  User |
| ------ | ------- | --------- | -------- | ------- | --------- | -------- |
| 3.8.1 | 4.xx | 0 | 0 | 0 | 0 | Average |
| 3.9.1-BETA-3965 worse | 4.xx | 0 | 0 | 0  | 0 | Average |
| 3.9.1-BETA-3965 best | 4.xx | 0 | 0 | 0  | 0 | Average |
|  |  | |  |  |  |  |  |  |  |

3d-gussner, 2020-07-10:
I hope some can make better pictures than I can to show off the results (good and bad).

carlin57, 2020-07-12:
Hello here are some tests with 3 filaments of different color and qualities. The filaments have variations in diameter depending on their quality ranging from 0.2mm of diameter variation to 0.8. For testing I used a mk3S with bondtech, it may be because of this that I see little or little diagonal ( crosses and rounds are landmarks for me). Print temperature 270 degrees.

To see the zig zag from front and behind I looked through the light as in the  pictures

[image]
[image]

First tests with the FW la15_acc_untested:

[image]

Then the FW 3.8.1:
[image]

And finally the FW 3.9.1:
[image]

Lots of waste lol:
[image]

For the results it's really a personal opinion:

FW la15_acc_untested:  E.jerk: 4.48 - 4.49
FW 3.8.1:  E.jerk: 4.5
FW 3.9.1 BETA 4202: E.jerk: 4.51

But for the three FW the visual zig zag is less good at the beginning and towards the end more numerous in zig zag and compact (personal opinion). On the sides the zig zag we see them less transparently but more against the day, I do not see how I could rate them from 0 to 10.
To decide which E.jerk value is the most correct I based on the four sides, all that is below I give a degressive note until the first layer.The same from my value to the top.

the FW 3.9 seems better to me, can be try with larger variations, start at 4.40 up to 4.60 with variations of 0.2 .